probonopd/WirelessPrinting #139: Alternatives to Micro SD cards for caching print files

TheAssassin, 2020-02-25:
How about using one of those "memory modules" for caching the gcode files? There's no need for nonvolatile storage; if the module is reset the data has to be resent anyway, I guess.
Using a Micro SD via SPI is more expensive (well, the modules cost the same, but you need an SD card, and you don't want to use the cheapest), and I think most gcode shouldn't exceed 128MBit (using a board with e.g., a W25Q128).

Might become obsolete when #45 is implemented.

probonopd, 2020-02-26:
That's a great idea!

TheAssassin, 2020-02-26:
If you have suggestions about which memory modules to use, please post them here. As said, 128 MBit isn't too bad, but I've seen gcode which exceeds 16 MiB. (Of course, two could be combined, but then it gets expensive again).

probonopd, 2020-02-27:
Unfortunately I have no experience whatsoever with this kind of modules. Are there cheap, ready-to-buy ESP modules with such chips onboard? Otherwise a microSD shield may be the more pragmatic (although overkill) solution for "normal users"...

TheAssassin, 2020-02-28:
Price wise using an SD card can be very viable. Depending on how well the FTL works in them, they will last basically forever with this application. Cheap asian ones with imaginative brands might not last as long as the big vendors', but I doubt even they would ever fail.

A module with the chip I mentioned costs < 2 USD and uses SPI for communication, like the MicroSD stuff (MicroSD natively supports SPI according to Wikipedia, actually). I think in the cheapest shops you can find them for even less.

TheAssassin, 2020-02-28:
A note on longetivity: both MicroSD and volatile memory should reach long lifetimes in this application. The former uses nonvolatile storage which has drastically lower wear counts (amounts of writes before failure), but as they're pretty large, we can spread the writes in many many locations. The built-in FTL takes care of that.

The volatile storage modules don't provide as much size, but volatile memory generally have really large wear counts. You can rewrite a volatile cell way more often than a nonvolatile. ¹

Technically I consider both systems suitable for this purpose. It's also less about the price. Using volatile storage is just some interesting experiment, I would say.

probonopd, 2020-03-07:
Agree. One thing to consider, with the SD card we might add some web UI that would allow one to retrieve and reprint past prints, which can be very handy in some situations.