probonopd/WirelessPrinting #141: Compile Error: redefinition of 'class fs::File'

cpeuschel, 2020-02-27:
Hi, 
while compiling i got some error:

..src/FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'
 class File : public FatFile, public Stream {
       ^

probonopd, 2020-02-27:
Hi, did you install the exact same combination of versions and commits as we are using in the Travis CI build? Which operating system are you on?

TheAssassin, 2020-02-27:
It always makes sense. The platformio definitions are written very quickly, and they have the advantage that you can predefine everything you need (e.g., board definitions). Upload settings can always be altered by the user. You can also provide multiple environments that all use the same board but use e.g., different upload ports or speeds (e.g., one for USB/serial, one for OTA updating).

cpeuschel, 2020-02-28:
Hi,
im using MacOs with Arduino IDE. Maybe i got the error because im using the NodeMCU Board and not the D1 mini? But if i burn your release bin (ESP8266WirelessPrintAsync_d1_mini_bad8e51.bin) on the NodeMCU it works fine on my printer.

If i using Linux with the configuration in the travis ci i got follow error:
/root/Arduino/libraries/AsyncElegantOTA/src/AsyncElegantOTA.h:51:71: error: 'U_FS' was not declared in this scope
                         int cmd = (filename.indexOf("spiffs") > -1) ? U_FS : U_FLASH;

cpeuschel, 2020-02-28:
Hi, 
with Version 1.0.6 of AsyncElegantOTA i got the error "error: 'U_FS' was not declared"
if i downgrade the Version to 1.0.4 (git clone https://github.com/ayushsharma82/AsyncElegantOTA ; cd AsyncElegantOTA ; git reset --hard 5c42ca2 ) then the compiling success. Can you reproduce that?

probonopd, 2020-03-07:
Argh, typical Arduino version hell. I had to do a lot of experimentation until I had found the exact combination of versions and hashes that works for me. All I can say is that the exact build configuration in `.travis.yml` works for me (and I am really hesitant to touch it).

TheAssassin, 2020-03-07:
@probonopd you should really prefer to use platformio instead of this Arduino IDE CLI compilation hell. With platformio, there's a very straightforward way to use libraries with specific versions. You can pin dependencies to repository tags or single commits, whatever is required.

TheAssassin, 2020-03-07:
@cpeuschel I could reproduce your issue perfectly while creating #144. I've pinned a few more libraries and frameworks to static versions and it builds perfectly now with platformio. Hopefully will be merged soon-ish.

cpeuschel, 2020-03-07:
@TheAssassin Build (nodemcuv2) was successful with vscode :)
Flash esp8266 with this command > esptool.py write_flash -fm qio 0x00000 firmware.bin

closed 2020-03-07 by cpeuschel

probonopd, 2020-03-07:
Thanks @TheAssassin you have made at least 2 people happy today