probonopd/WirelessPrinting #142: Printer won't start printing with Cura

cpeuschel, 2020-02-29:
im using Cura 4.5 and OctoPrint Version 3.5.12 on Mac OS Catalina (10.15.3) and the Continuous build on my esp8266. 
if i click "print with OctoPrint", the gcode is uploading and the LCD says "Receiving..." while the upload is done it says "Received" and than nothing happens.

Telnet Output: 
>M105
<ok T:20.62 /0.00 B:32.47 /0.00 @:0 B@:0#ok#
>M117 Receiving...
<ok#ok#
>M117 Receiving...
<ok#ok#
>M105
<ok T:20.62 /0.00 B:29.80 /0.00 @:0 B@:0#ok#
>M105
<ok T:20.62 /0.00 B:29.32 /0.00 @:0 B@:0#ok#
>M105
<ok T:20.62 /0.00 B:27.39 /0.00 @:0 B@:0#ok#
>M105
<ok T:20.62 /0.00 B:28.84 /0.00 @:0 B@:0#ok#
>M105
<ok T:20.62 /0.00 B:29.80 /0.00 @:0 B@:0#ok#
>M105
<ok T:20.62 /0.00 B:27.87 /0.00 @:0 B@:0#ok#
>M117 Received
<ok#ok#
>M105
<ok T:20.62 /0.00 B:27.87 /0.00 @:0 B@:0#ok#

if i start the print directly on "http://the-ip-address/" the printer starts the print.

probonopd, 2020-03-07:
I suspect Cura is not sending the `print` parameter:

https://github.com/probonopd/WirelessPrinting/blob/bad8e5166a2f1bbad95a4fb8849b08d1edfc7d3a/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino#L685-L692

Can you find a way to see the request that Cura is sending?

probonopd, 2020-03-07:
> OctoPrint Version 3.5.12

WirelessPrinting does not use nor need OctoPrint; it is a smaller implementation of it that goes on the ESP8266.

Which Cura plugin and version do you use to send the gcode to the printer?

cpeuschel, 2020-03-07:
> Which Cura plugin and version do you use to send the gcode to the printer?
Im Using Cura 4.5 with the OctoPrint Plugin Version 3.5.12

cpeuschel, 2020-03-07:
> Can you find a way to see the request that Cura is sending?

Maybe with Wireshark. I will check this

probonopd, 2020-03-07:
> Im Using Cura 4.5 with the OctoPrint Connection Plugin Version 3.5.12

Ah ok, somehow I was under the impression that one always needs a version of Cura-OctoPrintPlugin that matches Cura, but apparently this is (no longer?) the case.

cpeuschel, 2020-03-07:
Cura doesn't send the POST parameter "print", but why?

probonopd, 2020-03-07:
Great question. I am trying to reproduce it.

probonopd, 2020-03-07:
I can reproduce the issue with  Cura 4.5 and the OctoPrint Connection Plugin Version 3.5.12.

It also gives me this:

[image]

cpeuschel, 2020-03-07:
do u have discord? then we can debug this Problem and post the answer as a comment if we solved the Problem.

probonopd, 2020-03-07:
Same issue with Cura 4.4.1 and the OctoPrint Connection Plugin Version 3.5.12.
I suspect something in the OctoPrint Connection Plugin has changed.

probonopd, 2020-03-07:
* Power cycle printer if stuck on "Received"
* Close Cura

```
rm -rf ~/.local/share/cura/4.5/plugins/OctoPrintPlugin/*`
cd ~/.local/share/cura/4.5/plugins/OctoPrintPlugin
git clone https://github.com/fieldOfView/Cura-OctoPrintPlugin
cd Cura-OctoPrintPlugin/
git checkout 3cef0a9
```
* Restart Cura

__Works__

probonopd, 2020-03-07:
Can you post the POST request from a working and a non-working version for comparison?

probonopd, 2020-03-07:
Here something was changed regarding the print parameter:

https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/ab40c033cb36115575380c234340b893df091aad

probonopd, 2020-03-07:
I have opened tickets in the https://github.com/fieldOfView/Cura-OctoPrintPlugin project, let's continue there.

probonopd, 2020-03-07:
According to https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/156#issuecomment-596106186 it looks like we either need to support that additional "start printing" command, or just comment out the `if (request->hasParam("print", true))` condition so that we always start printing regardless of whether the client requests it or not.

cpeuschel, 2020-03-07:
Maybe we have to develop an Cura Plugin?

probonopd, 2020-03-07:
Fixed in the latest build. Thanks for reporting @cpeuschel

closed 2020-03-07 by probonopd