probonopd/WirelessPrinting #143: Octoprint error Forbidden

insidecircuits, 2020-03-04:
I get this issue lately, the plugin loses conection with octoprint.
When I go to manage printers -> connect octoprint there is a wrong API code. Even when I press request it gets the wrong API code from the webpage.
If I copy the corect code manually and hit enter, it also gets changed to the wrong one.
I even generated new API code and still does not work.

I have this in the log, repeating:

> 2020-03-04 20:55:23,434 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [956]: OctoPrintOutputDevice got an error while accessing <built-in method toString of QUrl object at 0x60c0006dbb48>
> 2020-03-04 20:55:23,434 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [957]: Forbidden
> 2020-03-04 20:55:33,403 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [719]: Received an unexpected returncode: 403
> 2020-03-04 20:55:33,406 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [956]: OctoPrintOutputDevice got an error while accessing <built-in method toString of QUrl object at 0x6080018c78d8>
> 2020-03-04 20:55:33,407 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [957]: Forbidden
> 2020-03-04 20:55:33,441 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [956]: OctoPrintOutputDevice got an error while accessing <built-in method toString of QUrl object at 0x6080018c6f38>
> 2020-03-04 20:55:33,441 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [957]: Forbidden

closed 2020-03-06 by insidecircuits

probonopd, 2020-03-06:
What was the resolution @insidecircuits? In case someone else runs into it.

insidecircuits, 2020-03-07:
I am sorry, I closed this because I mistakenly opened it here instead of in the OctoPrint plugin github issues.
For the resolution was to disable "Automatic discovery of octoprint instances" and add the API key manually and save.

probonopd, 2020-03-07:
Possibly this gets fixed by https://github.com/probonopd/WirelessPrinting/commit/a60dbef778705b2a8e911ff4933807dbbc6637b8.