probonopd/WirelessPrinting #148: Upload with Cura slow

cpeuschel, 2020-03-07:
is this normal, that the upload "sending date to OctoPrint" slow is? im uploading a gcode with 6,2 MB and it take over 10 minutes.

probonopd, 2020-03-08:
If you can upload your gcode somewhere, I can test it on my hardware.

cpeuschel, 2020-03-08:
https://we.tl/t-rHWLrZbrpg

probonopd, 2020-03-08:
Took 47 seconds on an ESP8266 with WirelessPrinting bad8e51, Cura 4.5, the latest OctoPrint plugin for Cura,  and a microSD card. I think this is about what you can expect from this type of hardware.

probonopd, 2020-03-08:
You can also test like this, without Cura:

```
ubuntu@ubuntu:~$ time curl -F "file=@/home/ubuntu/Downloads/CE2_03_Big_LED_Lampe_esp.gcode" -F "print=true" 192.168.0.24/api/files/local
{
  "files": {
    "local": {
      "name": "CE2_03_Big_LED_Lampe_esp.gcode",
      "origin": "local"
    }
  },
  "done": true
}
real	0m45.146s
user	0m0.023s
sys	0m0.031s
```

probonopd, 2020-03-08:
Using the precompiled WirelessPrinting c34dd15 from the GitHub Releases page, without microSD card, takes _forever_ because the SPIFFS area on the ESP8266 is not large enough for this file to work without an SD card.

cpeuschel, 2020-03-08:
> the SPIFFS area on the ESP8266 is not large enough for this file to work without an SD card.

ok this is my Problem. I need a SD-card shield. Who can read is clearly at an advantage :D

closed 2020-03-08 by cpeuschel