probonopd/WirelessPrinting #150: Printer stop the print randomly

cpeuschel, 2020-03-15:
if i start printing with Cura (4.5) with Octoprint-Cura-Plugin (3.5.12) and Continuous build (bf50a18) the printer stop the print randomly. 
Im using Wemos d1 mini with SD-Card shield
Any Ideas?

probonopd, 2020-03-15:
Can you see via telnet what is going on?

probonopd, 2020-03-15:
Do you have the same issue with this build? https://github.com/probonopd/WirelessPrinting/releases/tag/bad8e51

cpeuschel, 2020-03-16:
> Can you see via telnet what is going on?

Nothing unusual.

>Do you have the same issue with this build? https://github.com/probonopd/WirelessPrinting/releases/tag/bad8e51

Yes, same issue. 

It is necessary to power on the PC? i think if my mac goes to sleepmode, the print is stoping.

cpeuschel, 2020-03-16:
Ok, thats it. If i open an terminal with telnet an my macbook goes to sleep mode or i hit the button "go to sleep mode", the printer stops the print. If i start my mac again, the print go on.

probonopd, 2020-03-16:
> It is necessary to power on the PC? i think if my mac goes to sleepmode, the print is stoping.

Not for normal operation. With telnet, maybe.

probonopd, 2020-03-16:
Which kind of printer with which kind of mainboard and firmware are you using?

cpeuschel, 2020-03-16:
>Which kind of printer with which kind of mainboard and firmware are you using?

Ender2 with SKR mini E3 v1.2 with Marlin 2.0.4.4

cpeuschel, 2020-03-16:
maybe its not an issue

probonopd, 2020-03-16:
Could you try e.g., on the Ender-2 with Marlin 1.x?

cpeuschel, 2020-03-16:
Sure. It will take some minutes :D

cpeuschel, 2020-03-16:
I think i cant install marlin 1.x on the skr mini e3

probonopd, 2020-03-16:
ah, you don't have the original Ender-2 board left?

cpeuschel, 2020-03-16:
i dont think so. There no rx + tx pins left

cpeuschel, 2020-03-17:
With telnet connection it took 10 secounds to write "sending"
[image]

With telnet connection and sleep mode by mac it took over one minute.
[image]

cpeuschel, 2020-03-17:
I have modify follow code:
inline void telnetSend(const String line) {
  Serial.println("\n sending");
  if (serverClient && serverClient.connected())     // send data to telnet client if connected
  {
    Serial.println("client available");
    serverClient.println(line);
    Serial.println("sending");
  } else {
    Serial.println("Client not available");
  }
}

probonopd, 2020-03-17:
What is that line of ?=)(/&$§"§$%()=)&$§$%/)= there, do you have any clue where this might be coming from?

cpeuschel, 2020-03-17:
no, sorry. I will try to find out. 

BTW. can u reproduce that?

probonopd, 2020-03-17:
When does this ?=)(/&$§"§$%()=)&$§$%/) occur: Immediately after you connect using telnet or when you send a print job?

cpeuschel, 2020-03-17:
i see this if i connect the d1 to Serial-Monitor without printer is connected

probonopd, 2020-03-17:
Please describe your test setup more completely. What have you connected to what, and what do you do when the  ?=)(/&$§"§$%()=)&$§$%/) occurs? Currently I cannot reproduce your results.

cpeuschel, 2020-03-17:
if i connected the wemos d1 with an usb cable to my Mac (or Windows); open the Serial-Monitor with the Arduino IDE (Baudrate 115200) and the "?=)(/&$§"§$%()=)&$§$%/)" occurs.

probonopd, 2020-03-17:
You need to connect the wemos d1 mini to your 3D printer board using the pinout described. Then connect to the wemos d1 mini from your computer via telnet. Everything else won't work because the code will not run if no printer board is attached that responds to the commands...

cpeuschel, 2020-03-17:
Ich muss mal deutsch schreiben, weil mein englisch so lala ist, wie du vielleicht gemerkt hast. :)

Mir ist klar, dass ich den D1 mit meinem Drucker verbinden muss, um überhaupt drucken zu können.

Ich habe dieses Setup nur aufgebaut um zu prüfen, warum mein Drucker nicht weiter druckt wenn mein Macbook in den Ruhezustand geht und ich via Telnet dran bin.

Generell ist mir das (https://github.com/probonopd/WirelessPrinting/issues/150#issuecomment-600216686) aufgefallen.

probonopd, 2020-03-17:
Das Telnet braucht man ja nur zum Debugging, und während der Zeit sollte das Macbook so eingestellt werden dass es nicht in den Ruhezustand geht.

cpeuschel, 2020-03-17:
Ja gut. Aber Strange ist es trotzdem. Lass uns hier schließen.

probonopd, 2020-03-17:
Please add a comment here if you are experience sudden print stops, too. Then we will reopen the ticket.

probonopd, 2020-06-02:
I experienced this once (WirelessPrinting bad8e51 on Prusa i3).

Need to find out whether this is also a problem in https://github.com/probonopd/WirelessPrinting/releases/tag/bad8e51 or "only" in the continuous build.

probonopd, 2020-06-02:
We should come up with some large test gcode that we can run on printers without needing any filament... to find out what is going on.

baykah, 2020-09-03:
it did it again today, 1mo gcode in spiff, print stoped at 85% saying complete and when I try to dl the gcode back, stop at 85% ..

I just received a wemos and the SD shield to try, could it be that it is link to upload speed / time ?, 1mo gcode seemed very long ..

It looks like the file may be corrupted in the flash ?

probonopd, 2020-09-03:
Unfortunately I don't know what is causing this yet.

baykah, 2021-01-23:
Hello ! i'm back, sorry I completely stopped using your solution, because with the self complited modified prusa firmware to use serial 2, I started having issue even with printing from SD, no bed heating, hang up of the printer. I went back to official .bin from prusa and everything went back to normal.

Then a few month ago, I bought a daugter board with TMC sillent driver that need to solder wire in the back of the rambo mini board.

Then, today, I dismounted the board again to solder wire to TX and RX test pad in the back of the board, in the mean time, I got some WEMOS D1 with the SD shield.

I'll start again experimenting your work ! but with the "original" firmware from prusa

probonopd, 2022-04-26:
Could it be that the WS2812 neopixel code is causing this?

baykah, 2022-04-28:
zero issue so far since moving to original prusa firmware and to WEMOS D1 with the SD shield

probonopd, 2022-08-24:
In case this issue arises again, for debugging/prolonged stress testing without needing to actually print anything:

> Marlin includes a native simulator that runs on Windows, Linux, and macOS, an invaluable tool for speedier development and debugging. It can simulate either a common Graphical LCD or TFT with Classic UI.

https://twitter.com/MarlinFirmware/status/1561811939642179586

> 1. Download the "Simulator" example configuration which uses motherboard BOARD_LINUX_RAMPS.
2. Build the firmware most easily with the "Build" button in the Auto Build Marlin extension for VSCode.
3. Click the "Run" button or reveal and double-click the built native binary.

https://twitter.com/MarlinFirmware/status/1561813417190735873

Possibly this **Marlin timing simulator** is even more helpful?

https://github.com/eyal0/Marlin