probonopd/WirelessPrinting #153: Crash when receive job

romandoss, 2020-04-11:
I try to send some job(Cura 4.5 or PrusaSlicer 2.2), but esp32 always crash with this error:

E (13543) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13543) task_wdt:  - async_tcp (CPU 1)
E (13543) task_wdt: Tasks currently running:
E (13543) task_wdt: CPU 0: IDLE0
E (13543) task_wdt: CPU 1: loopTask
E (13543) task_wdt: Aborting.
abort() was called at PC 0x400ed0eb on core 0

Backtrace: 0x4008c674:0x3ffbe170 0x4008c8a5:0x3ffbe190 0x400ed0eb:0x3ffbe1b0 0x400849b1:0x3ffbe1d0 0x4016ccdf:0x3ffbc270 0x400ee4a2:0x3ffbc290 0x4008a5a1:0x3ffbc2b0 0x40088dbd:0x3ffbc2d0

Rebooting...

E (714631) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (714631) task_wdt:  - IDLE0 (CPU 0)
E (714631) task_wdt: Tasks currently running:
E (714631) task_wdt: CPU 0: async_tcp
E (714631) task_wdt: CPU 1: loopTask
E (714631) task_wdt: Aborting.
abort() was called at PC 0x400ed0eb on core 0

Backtrace: 0x4008c674:0x3ffbe170 0x4008c8a5:0x3ffbe190 0x400ed0eb:0x3ffbe1b0 0x400849b1:0x3ffbe1d0 0x40081c0b:0x3ffd8a40 0x400828e7:0x3ffd8a60 0x400811f3:0x3ffd8a80 0x400e7175:0x3ffd8aa0 0x400e71d9:0x3ffd8ac0 0x400e7657:0x3ffd8ae0 0x400e784e:0x3ffd8b20 0x4010bc9e:0x3ffd8b50 0x4010a10a:0x3ffd8b70 0x4010b6a5:0x3ffd8bb0 0x4012728a:0x3ffd8be0 0x4000bd83:0x3ffd8c00 0x4000117d:0x3ffd8c20 0x40058af1:0x3ffd8c40 0x4016bff9:0x3ffd8c70 0x4016c045:0x3ffd8cb0 0x4015e6ba:0x3ffd8cd0 0x4016c425:0x3ffd8cf0 0x400d8095:0x3ffd8d10 0x400d648f:0x3ffd8d30 0x400d2231:0x3ffd8dc0 0x4016c963:0x3ffd8e00 0x4016c7fa:0x3ffd8e50 0x400dc325:0x3ffd8e80 0x400ddd8a:0x3ffd8ef0 0x400ddf65:0x3ffd8f30 0x4015d9c5:0x3ffd8f50 0x4015da55:0x3ffd8f90 0x4015e102:0x3ffd8fb0 0x40088dbd:0x3ffd8fe0

probonopd, 2020-04-11:
Does the same happen if you upload gcode using the web interface?
Do you have an ESP8266 and can test if it happens there too?

romandoss, 2020-04-11:
Same, web page goes offline then esp32 reboots but have no message in console.
And all of this is happen when 3d printer is not connected! When I connect them both, all worked as expected.

probonopd, 2020-04-11:
> And all of this is happen when 3d printer is not connected! When I connect them both, all worked as expected.

The sketch does not work without a printer attached to the ESP board. Only ever run/test the sketch with a ESP board attached.

closed 2020-04-11 by probonopd