probonopd/WirelessPrinting #156: [Recommendation] README.md; Comparisons Great, but how does it compare with ESP3D ?

borland1, 2020-05-04:
Great comparision between SD-SLOT/Octoprint/WirelessPrint, but what is the difference between ESP3D which seems to be similar or maybe have more features?

ESP3D development seems to still be active.  Is that the case with WirelessPrint?

probonopd, 2020-05-05:
This discussion caused WirelessPrinting to be made:

* https://github.com/luc-github/ESP3D/issues/149
* https://github.com/luc-github/ESP3D/issues/151

Can you send a suggestion for a text that cold be added to the README based on this?

WirelessPrinting is active, it does what I need. If someone needs it to do more, then they are welcome to send pull requests.

borland1, 2020-05-09:
Having only browsed these product's documents, here's my take.
[image]

[SerialWiFiHostAlternatives.zip](https://github.com/probonopd/WirelessPrinting/files/4602385/SerialWiFiHostAlternatives.zip)

probonopd, 2020-05-09:
Well, I think the main difference is that ESP3D doesn't have the main feature of either WirelessPrint or OctoPrint, which is: Computer can be turned off after having sent off a print job to the printer. So it does something entirely else. Or has this changed recently?

borland1, 2020-05-09:
Don't know what ESP3D is working on but PC's today are multitasking.  PC OS power saving features should be enabled if that's a concern.

As for printer power saving features upon print completion, Marlin has that supported ( if enabled in config.h ) with g-codes M80 (power supply ON) and M81 (power supply OFF).  Originally for ATX power supply hardware, but hardware is not mandatory to use M81 power supply feature.  ATX power OFF is not completely OFF, but allows for at least 300mA supply for low current needs (such as the main microprocessor).  M81 g-code command provides for:
- disabling all bed/extruder heaters, waiting for any movement command completion, then - disabling all stepper motor currents,
- disabling extruder fan(s)

Besides adding M81 to Slicer's print completion g-code (end-code), might also include moving print head to a parked position.

closed 2020-05-10 by borland1

probonopd, 2020-05-10:
Are you running your PC all the time while doing a 24-hour print? I don't. Besides, I like to reboot my PC much more frequently than my prints take.