probonopd/WirelessPrinting #157: Test with Prusa Mini

probonopd, 2020-06-08:
The Prusa Mini has a slot for an ESP-01 module.

Can we get WirelessPrinting to run on it?

[image]

Prusa Mini owners are asked to try and report, since I do not have the printer.

cpeuschel, 2020-07-12:
Hi @probonopd,

my Prusa Mini is arrived a few days ago. So i will try and report tomorrow.

cpeuschel, 2020-07-13:
Hi,

Prusa Mini Firmware: [4.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.1.0)
D1 mini Firmware: [acaed49](https://github.com/probonopd/WirelessPrinting/commit/acaed497142244f7e95d25bc3a13e4a6f1c886a0)

The d1 can not connect to the printer. here is the output from telnet:

`Connected to esp_d40674.
Escape character is '^]'.
Connecting at 115200
>M115
<�M115#discovering#
<e3#discovering#
Connecting at 250000
>M115
Connecting at 250000
>M115
Connecting at 57600
>M115
Connecting at 57600
>M115
Connecting at 115200
>M115
Connecting at 115200
>M115
<�M115#discovering#
<e3#discovering#
`

i have no clue why it says "discovering".

probonopd, 2020-07-23:
Thanks for testing @cpeuschel. Would have been too easy if it "just worked"...

cpeuschel, 2020-10-02:
if you would connect Octoprint you must connect the pi via micro USB cable into the printer.

So if we connect the wemos d1 with an micro USB cable directly on the printer maybe it work? Provided there is a way to send the commands (M115, etc.) via USB? @probonopd is that possible?

probonopd, 2020-10-03:
Not with just the ESP8266 or ESP32. For this we would need a device with USB host capabilities.

BluPix, 2020-10-17:
Hi, I have flashed ESP-01 to your firmware, it worked on the programmer but It didn't work in the Prusa buddy (clone). I have seen in router logs that esp acquired IP, but it didn't communicate from browser. It looks like the actual Prusa firmware set reset pin to turn off ESP.

probonopd, 2020-10-17:
Interesting. Thanks for testing. Can you connect to the ESP via `telnet` and see if there are any messages there?

vilhelmen, 2020-11-28:
Hi there, I flashed `ESP8266WirelessPrintAsync_d1_mini_acaed49.bin` to my ESP8266 and installed it into my mini running `4.2.1-final`. I can somewhat confirm what BluPix says, the website does not appear to be loading now that I've installed it, but I have confirmed it acquired a DHCP lease on boot. There don't appear to be any open ports, so it may be going back offline.

I only have a multimeter so I can't do too much diagnostics, but post-boot it appears that IO0, IO2, and EN are high and RST is low.

vilhelmen, 2020-11-28:
I built a serial mitm device out of two usb dongles and after grounding RST momentarily it seems to just work, even after switching it off and unplugging it, weirdly enough. Not sure if I broke it or what. Maybe it had it seated poorly? In order to install it I have to take out the power switch and it's a pain in the butt, so I don't want to do that until I know it's working.

I wasn't able to send a large gcode file through (30MB), though I was (seemingly) able to upload a small (830K) sample file with curl that came with the printer. I wasn't able to print or download the uploaded file. I uploaded a text file with a few characters in it and still didn't seem to be able to download it. I think this thing only has 1M of flash, so that could be part of the issue, I'm not sure.

If I fiddle with socat more I may be able to get hex dumps of the communication if you have things you want to try.

probonopd, 2020-11-29:
Great @vilhelmen, thanks for testing.

When it is "half-working", can you connect to the ESP with telnet, to see any debug messages?

It is expected that this currently works only for gcode files that are small enough to fit into the internal memory of the ESP chip. For larger files, you could use a different ESP module with a SD card slot and a microSD card attached to the ESP.

I guess this is not what Prusa has in mind with the Prusa mini connector for the ESP01. I guess they will make a firmware that will allow the ESP01 to upload files to the microSD card attached to the printer. But that needs to be implemented in the Prusa firmware. Until they do that, have fun tinkering with WirelessPrinting!

vilhelmen, 2020-11-29:
I completely forgot about the telnet port. Now that I'm working on it today, the serial line is completely silent. Telnet connects on boot, but remains quiet and a reconnect fails. RST to GND is reporting -40mV, which seems suspicious.

Connecting/disconnecting RST causes a bunch of data to be output on both lines, but that could be noise, and seems to eventually trigger a reset. The mini starts emitting the `4d 31 31 35 0a 65 33 0a` heartbeat/discovery/whatever I saw the day before. The ESP's page is up, but telnet appears to suggest it's stuck in baud rate discovery. The telnet logs say it's receiving the mini's message, but nothing is coming out of the esp today.

The mini, as far as I can tell, doesn't have a microSD slot. There's a mystery barrel plug(?), some sort of small usb, I never remember which is which, a USB A port, and ethernet. There's a nonzero chance I broke something, and my setup is a mess of wires, so something could have come loose. I'll rewire and see what I can come up with, but no promises.

probonopd, 2020-11-29:
> The mini, as far as I can tell, doesn't have a microSD slot. 

Ah, right. I wonder where Prusa wants to buffer large files then? USB? How does it work for wired Ethernet?

vilhelmen, 2020-11-29:
That's a good question! Did some light googling and it seems to suggest that you just don't, it's monitor only.

https://forum.prusaprinters.org/forum/general-discussion-announcements-and-releases/can-you-do-anything-with-the-mini-over-ethernet/

https://help.prusa3d.com/en/article/prusa-connect-local-mini_133279

probonopd, 2020-11-29:
What? You can't send a Gcode file to a Prusa Mini over Ethernet? That'd I'd call a fail then...

vilhelmen, 2020-11-29:
Yeah, this is getting more disappointing the more I look into it :/

vilhelmen, 2020-11-30:
Well, I rewired everything and I'm powering the ESP externally. If I only connect the serial lines, the ESP doesn't get past the "Connecting at X" loop. There's a response from the printer after the 115200 pings, but it's inconsistent. Nothing like the consistent `4d 31 31 35 0a 65 33 0a` I got before.

Maybe I still have a faulty connection, or maybe I shorted something in my < 1 week old mini, hard to tell.

Anyeos, 2021-02-12:
> Well, I rewired everything and I'm powering the ESP externally. If I only connect the serial lines, the ESP doesn't get past the "Connecting at X" loop. There's a response from the printer after the [...]

Have you resolved your problem? The TX and RX lines are some "resistant" because they only pull down. If you connet by mystake to GND there are no damage because the HIGH state is achieved with a pull up resistor. And if you connet to Vcc the circuit is more tolerant as I know. 
So it is likely a out of sync problem. Try with the shortest wires you can.
And look at the datasheet of the SoC that is on your mainboard (of your printer). There are some SoC (MCU = Microcontroller Unit) that runs on some clock frequency that does not achieve exactly 115200 bps but 115384. So that can raise to communication problems.
If you can compile WirelessPrinting and knows exactly what BPS are using your printer try to specify that exact bps on WirelessPrinting.
But better if you can achieve the same exact bps on both devices.

fgrizelj, 2021-02-22:
> if you would connect Octoprint you must connect the pi via micro USB cable into the printer.
> 
> So if we connect the wemos d1 with an micro USB cable directly on the printer maybe it work? Provided there is a way to send the commands (M115, etc.) via USB? @probonopd is that possible?

Ive been investigating this topic last few days, and also came to a conclusion that rx and tx on esp cannot be used since there is nk firmware support for it. Since comm via usb seems to work fine it also crossed my mind that i should look into a board wgich has a wifi module and usb. But looking at wemos specs it seems to me that usb is used for programming and power. Also the whole board seems to be powered by 5V. Prusa mini only has 3.3V ehich are added for esp support.

probonopd, 2021-02-22:
A 3D printer with no support for RX and TX?

fgrizelj, 2021-02-22:
I am new to this whole scene, but thats what it looks like to me. They seem to be trying to add it:

https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1246

And one other comment also caught my eye:

https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1246#issuecomment-774498931

Turro75, 2021-03-15:
This is the changes I made to prusa mini source in order to get the uart6 working, it works as I'm using an rpi zero powered directly from esp port present on buddy.

the main issue I found while trying esp3d is the lack of sd support in the firmware.
[myedit.diff.zip](https://github.com/probonopd/WirelessPrinting/files/6144113/myedit.diff.zip)

probonopd, 2021-03-15:
@Turro75 if you can print on the Prusa Mini over serial with a Raspberry Pi, then you might be able to print using WirelessPrinting as well. No special support in the firmware is needed apart from the serial connection.

Turro75, 2021-03-15:
Ouch...I haven't check where I am...
I'll try wireless printing soon and I'll let You know

cpeuschel, 2021-07-15:
@Turro75 can you upload your modified Firmware?

Turro75, 2021-07-17:
[firmware.bbf.zip](https://github.com/probonopd/WirelessPrinting/files/6834523/firmware.bbf.zip)

this is a firmware based on Llama custom firmware with uart6 enabled.

here my pull request that generated the bbf file:
[https://github.com/matthewlloyd/Llama-Mini-Firmware/pull/7](https://github.com/matthewlloyd/Llama-Mini-Firmware/pull/7)

cpeuschel, 2021-07-18:
Connect from d1 to the mini seems work.

[image]

Prusa Mini connection
[image]

Prusa Mini Firmware
[image]

i will tests some prints soon.

probonopd, 2021-07-18:
@cpeuschel that looks promising!
I wonder though, isn't the Prusa Mini supposed to have a slot for a ESP-01 module?

cpeuschel, 2021-07-18:
With the ESP-01 module you can't upload large gcode files. For this reason I used a wemos d1 mini

BluPix, 2021-07-21:
@Turro75 thanks for your firmware, I had to add line 
HAL_GPIO_WritePin(GPIOC, ESP_RST_Pin, GPIO_PIN_SET);
to src/main.cpp under line 935 for ESP-01 mod start. It didn't print yet, WirelessPrinting is trying all baudrates and nothing responds.

Turro75, 2021-07-21:
before inserting it in the slot I made a lead jumper across pins 3.3 - RST - CHPD.
otherwise a reset logic must be handled by firmware.
[image]

Emanuelgtxf, 2021-11-30:
Boas eu na entendo nada disto mas o que reparei antes de instalar o esp01 é que tem 1 conector de antena logo o lado ! Pode estar relacionado?

foobarjimmy, 2021-12-08:
Did you use that uart6 enabled firmware or official firmware?