probonopd/WirelessPrinting #160: Include relevant libraries

jkruiktech, 2020-11-14:
It would be good to have a version that doesn't assume any additional libraries are already installed. Maybe I should put that together in a separate project? (once I have all my dependencies sorted out myself).
Might help some people to get off the ground a bit quicker, especially those who aren't quick at debugging (e.g. me)

probonopd, 2020-11-14:
Are you building like this?

```
# Get PlatformIO (the toolchain we use for compiling)
git clone https://github.com/probonopd/WirelessPrinting
cd WirelessPrinting
wget -c https://downloads.egenix.com/python/install-pyrun
bash install-pyrun --python=3.5 pyrun/
pyrun/bin/pip3 install -U platformio==4.2.1

# Build the firmware (it downloads the needed libraries)
pyrun/bin/platformio run
find . -name "*firmware.bin"
```

jkruiktech, 2020-11-16:
No sorry I hadn't been building using platformIO. That would be the issue, thanks!

closed 2020-11-16 by jkruiktech