probonopd/WirelessPrinting #161: SDfat compile error

wjflier, 2021-01-27:
can't get past this error. " \Documents\Arduino\libraries\SdFat-2.0.4\src/SdFat.h:423:16: error: 'typedef class File32 File' conflicts with a previous declaration"

probonopd, 2021-01-27:
Probably you are using some different revision of the library than we used?

wjflier, 2021-01-27:
That's what I thought, tried a couple.  Guess it's hit and miss

probonopd, 2021-01-27:
Best start with the exact combination of revisions we are using, and then go from there.
The are automatically used by Platformio but also documented here:

https://github.com/probonopd/WirelessPrinting/blob/acaed497142244f7e95d25bc3a13e4a6f1c886a0/platformio.ini#L19-L26

wjflier, 2021-01-27:
Ok, I'll play with that.  Thanks

wjflier, 2021-01-27:
No joy,  same error

wjflier, 2021-01-29:
I changed all LIbs. to no avail.  unfortunately most of them have been updated.  I never got Platformio to work on my pc.  So as of now it seems this ino is a no go.

probonopd, 2021-01-30:
What IDE are you using? PlatformIO? If not, please retry with it.

wjflier, 2021-01-30:
Thanks for the reply.  I using Arduino 1.8.9. Tried 1.8.7 too.  Can't get PlatformIO to run on my pc.  Tried several FS.h files.  Looks like #define FS_NO_GLOBALS is not doing it's job.  Probably move to my new pc this weekend. Win10.

probonopd, 2021-01-30:
OK, we have never tested this on Windows. Best use PlatformIO on a Linux system that supports it.

wjflier, 2021-01-30:
Will do, thanks.

wjflier, 2021-02-03:
FYI, flashed with Pyflasher, worked like a charm. next a good HTML interface.

closed 2022-04-26 by probonopd