probonopd/WirelessPrinting #162: multiple definition of...

Anyeos, 2021-02-12:
I cannot link, there are a lot of redefined functions.

I don't found information about it in the proyect and I already had installed PlatformIO and sucessfully compiles other proyects without issues. So it is something specific with this proyect in particular.

Version of compiler is: gcc/xtensa-lx106-elf/4.8.2
All updated.

The ESP32 version compiles OK but I want to compile the ESP8266 version:
nodemcuv2      FAILED    00:00:53.661
d1_mini        FAILED    00:00:04.210
esp32dev       SUCCESS   00:00:54.464

### SOLVED:
I realized that the problem was because the project is using two libraries with the same functions: 
- ESPAsyncTCP
- AsyncTCP

So, why it is using two different libraries that have the same functions?
I deleted AsyncTCP (from ".pio/libdeps" directory) and now compiles and link successfully.

I will leave this issue anyway so other people with the same problem can solve it.

probonopd, 2021-02-12:
Excellent question, I think one is for ESP32 and the other is for ESP8266 (as far as I remember).
Things may have improved with the libraries in the meantime, so maybe it could be done in a better way using never versions of the libraries.

Anyeos, 2021-02-12:
> Excellent question, I think one is for ESP32 and the other is for ESP8266 (as far as I remember).
> Things may have improved with the libraries in the meantime, so maybe it could be done in a better way using never versions of the libraries.

There are separate directories for each, so no reason to put two different libraries with the same functions on the same directory.
The directory structure is as follows:
```
└── libdeps
    ├── d1_mini
    │   ├── ArduinoJson
    │   ├── AsyncElegantOTA
    │   ├── ESP Async WebServer
    │   ├── ESPAsyncTCP
    │   ├── ESPAsyncTCP@src-66213e8e5399ee92a319916838e73378
    │   ├── ESPAsyncWifiManager
    │   ├── NeoPixelBus
    │   └── SdFat
    ├── esp32dev
    │   ├── ArduinoJson
    │   ├── AsyncElegantOTA
    │   ├── AsyncTCP
    │   ├── ESP Async WebServer
    │   ├── ESPAsyncTCP
    │   ├── ESPAsyncWifiManager
    │   ├── Hash
    │   ├── NeoPixelBus
    │   └── SdFat
    └── nodemcuv2
        ├── ArduinoJson
        ├── AsyncElegantOTA
        ├── ESP Async WebServer
        ├── ESPAsyncTCP
        ├── ESPAsyncTCP@src-66213e8e5399ee92a319916838e73378
        ├── ESPAsyncWifiManager
        ├── NeoPixelBus
        └── SdFat
```
As you can see there are one library group for each environment. ESP32 have it owns and so.
Well, I don't know, maybe some upgrading still in process.

Anyeos, 2021-02-15:
I forget to mention that I tested with ESPAsyncTCP and next with AsyncTCP for curiosity. And I noted that the ESPAsyncTCP is more faster than AsyncTCP. So my guessing is that the AsyncTCP is implemented purely in software while ESPAsyncTCP may be using some ESP internal functions.

I suggest to use ESPAsyncTCP first and if you encourer problems try with AsyncTCP.

Anyeos, 2021-02-20:
I found a "lawful" solution. There are other project using ESP8266 and platformIO and that project have the next lines on "platformio.ini":
```
lib_compat_mode = strict
lib_deps =
    fastled/FastLED @ 3.3.2
    NeoPixelBus @ 2.6.0
    ESPAsyncTCP @ 1.2.0
    ESPAsyncUDP
    AsyncTCP @ 1.0.3
    IRremoteESP8266 @ 2.7.3
    https://github.com/lorol/LITTLEFS.git
    https://github.com/Aircoookie/ESPAsyncWebServer.git @ ~2.0.0
  #For use of the TTGO T-Display ESP32 Module with integrated TFT display uncomment the following line
    #TFT_eSPI
  #For use SSD1306 OLED display uncomment following
    #U8g2@~2.27.2
  #For Dallas sensor uncomment following 2 lines
    #OneWire@~2.3.5
    #milesburton/DallasTemperature@^3.9.0
  #For BME280 sensor uncomment following
    #BME280@~3.0.0
lib_ignore =
    AsyncTCP
```
So, the "lawful" way to fix it is using the "lib_ignore" directive.

probonopd, 2021-02-20:
> "lib_ignore" directive

Thank you for finding this out!
Will try it out soon.

probonopd, 2022-04-26:
The builds are working again.

closed 2022-04-26 by probonopd