probonopd/WirelessPrinting #163: Files sent from Cura have the word "true" inside.

Anyeos, 2021-02-12:
When I send a file from the served webpage directly, it uploads correctly and have all the bytes inside (on the SDCard). But when I upload directly from Cura, the file size on the SD Card is only 4 bytes long and have the words "true" inside. So, all the file content is truncated with the word "true".
I think it is something related with Cura sending some command that WirelessPrinting missunderstands or maybe it is a bug.

Anyeos, 2021-02-12:
I discovered that the problem is in:
```
  if (!index) {
    lcd("Receiving...");
    ...
```
Becasue Cura send index = 0 sometimes. So I will implement some other manner of checking to receive a file.

Anyeos, 2021-02-12:
I think Cura is sending 3 portion of different content with the same filename.
I just put a counter on the filename so I get filename+counter and I recevied 3 files, the last two with "true" inside. The first one is the real file.
So I need to implement some code to handle that correctly.
I guess that one of the "true" is the order to print. But to be sure I need to check Cura documentation.

Anyeos, 2021-02-12:
Here a raw and rudimentary but working code for the purpose. Only modified the function "handleUpload".

```
int receivecount = 0;
String lastUploadedFullname;
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  static FileWrapper file;

  if (!index) {
    // No quiero borrar los archivos, los quiero conservar
    //if (uploadedFullname != "")
      //storageFS.remove(uploadedFullname);     // Remove previous file
    int pos = filename.lastIndexOf("/");
    uploadedFullname = pos == -1 ? "/" + filename : filename.substring(pos);
    if (uploadedFullname.length() > storageFS.getMaxPathLength())
      uploadedFullname = "/cached.gco";   // TODO maybe a different solution

    if (lastUploadedFullname != uploadedFullname) {
      receivecount = 0;
    } else
    if (receivecount >= 3) {
      receivecount = 0;
    }

    receivecount++;
    if (receivecount <= 1) {
      file = storageFS.open(uploadedFullname, "w"); // create or truncate file
      lastUploadedFullname = uploadedFullname;
      lcd("Receiving: "+uploadedFullname);
    }
  }

  if (receivecount > 1)
    return;

  file.write(data, len);

  if (final) { // upload finished
    file.close();
    uploadedFileSize = index + len;
  }
  else
    uploadedFileSize = 0;
}
```

probonopd, 2021-02-12:
Thank you @Anyeos. Do you think you could send a [pull request](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests)?

Anyeos, 2021-02-12:
> Thank you @Anyeos. Do you think you could send a [pull request](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests)?

I need to improve the code if not you will need to upload 3 times from the web to bypass the counter (that is some workaround for Cura). For now it is working as expected with Cura 4.8 only. I will improve it eventually but not today.
For now you can copy / paste the above code.

probonopd, 2022-04-26:
Does this also continue to work with PrusaSlicer @Anyeos?

Anyeos, 2022-05-06:
Hello, all is working good for now with Cura, PrusaSlicer, etc. I improved the code a lot so it is working better. Can cancel, send file, print, cancel, print... it works good.
It does not mean it is perfect but I am using it periodically without issues.

The code is published on a fork that I made. If you have the time to study the code feel free to do so. I am very busy and cannot do a pull request or a path.

probonopd, 2022-05-06:
That sounds interesting @Anyeos. Thanks for letting me know.

probonopd, 2024-11-23:
Hello @Anyeos, thanks for your work. How about sending pull requests so that your improvements can be upstreamed?