probonopd/WirelessPrinting #165: Not working with Marlin 2.0

BaltasarParreira, 2021-04-07:
Hi, I just compiled and uploaded to a nodemcu that I connect to my printer.
Comunication with the printer looks working, but GCODE commands are not 100% understanded and I got some errors so no printing on sending a file from Cura.

Here is what I can see from telnet:
```

>M105
<ok T:24.37 /0.00 B:29.23 /0.00 @:0 B@:0#ok#
>M117 Receiving...
<ok#ok#
>M105
<ok T:24.80 /0.00 B:29.03 /0.00 @:0 B@:0#ok#
>M105
<ok T:23.71 /0.00 B:29.35 /0.00 @:0 B@:0#ok#
>M105
<ok T:24.26 /0.00 B:29.03 /0.00 @:0 B@:0#ok#
>M105
<ok T:25.34 /0.00 B:28.38 /0.00 @:0 B@:0#ok#
>M105
<ok T:24.96 /0.00 B:27.76 /0.00 @:0 B@:0#ok#
>M117 Receiving...
<ok#ok#
>M117 Receiving...
<ok#ok#
>M117 Received
<ok#ok#
>M300 S500 P50
<ok#ok#
>M117 Printing...
<ok#ok#
>M300 S500 P50
<ok#ok#
>M105
<ok T:23.55 /0.00 B:28.32 /0.00 @:0 B@:0#ok#
>true
<echo:Unknown command: "true"#wait more#
<ok#ok#
>M117 Complete
<ok#ok#
>M105
<ok T:24.06 /0.00 B:28.84 /0.00 @:0 B@:0#ok#
>M105
<ok T:24.45 /0.00 B:28.78 /0.00 @:0 B@:0#ok#
>M105
<ok T:23.05 /0.00 B:27.87 /0.00 @:0 B@:0#ok#
>M105
<ok T:25.65 /0.00 B:29.06 /0.00 @:0 B@:0#ok#
>M105
<ok T:24.88 /0.00 B:27.90 /0.00 @:0 B@:0#ok#
>M105
```

Your "WirelessPrinting" approach looks amazing and simple for all I need.
I try so many other options and they never worked 100% for my requirements that is just send a file and print it !!!
If you want I can help making this working 100% with latest Marlin.
My hardware is a 32bit MKS Robin Nano 1.2 board with a 3,5" touch TFT and I'm using one of the board serial ports to connect with the nodemcu.

Thanks.

probonopd, 2021-04-08:
That `true` looks suspicious. Is it in the gcode?

BaltasarParreira, 2021-04-08:
Yes I'm checking the gcode list from marlin and a lot of codes you use they don't exist.
For example I don't even know if there is a way of direct print, looks like that the file needs to be stored first at the local board SD card and them issue the right gcodes to select it and start printing.
Your code needs a couple of changes to be 100% compatible, but it's doable as if OctoPrint works, this should wokr too.

closed 2022-04-26 by probonopd