probonopd/WirelessPrinting #169: about use serial0 on esp32

leoncoolmoon, 2021-11-13:
Appearently you can connect GPIO15 to GND to disable initial output from the ROM ootloader.
ref: https://community.platformio.org/t/esp32-prevent-from-sending-serial-output-at-startup/18756/5

probonopd, 2021-11-14:
Would you like to test it and send a pull request if it works? Thank you very much.

leoncoolmoon, 2021-11-17:
it is my first time trying this project.
I really like your idea.
I was trying to see how it work on esp8266.
when I build this project with "platformio==4.2.1"
I have this result.
```Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:16.742
d1_mini        FAILED    00:00:16.502
esp32dev       SUCCESS   00:00:17.254
============================================ 2 failed, 1 succeeded in 00:00:50.498 ============================================ 
The terminal process "C:\Users\lenovo\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```
seems that the code has some "multiple definition" issue for esp8266?

```Executing task: C:\Users\lenovo\.platformio\penv\Scripts\platformio.exe run --environment d1_mini <

Processing d1_mini (board: d1_mini; platform: espressif8266@2.0.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.0.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20500.190223 (2.5.0)
 - tool-esptool 1.413.0 (4.13)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting ESP8266WirelessPrintAsync.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SdFat> 1.1.0+sha.3b79f38
|   |-- <SPI> 1.0
|-- <ESP Async WebServer> 1.2.0+sha.95dedf7
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.9.1+sha.3df4efd5   
|-- <ESPAsyncTCP> 1.2.0+sha.7e9ed22        
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncElegantOTA> 1.0.3+sha.4b3528c
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP Async WebServer> 1.2.0+sha.95dedf7
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.9.1+sha.3df4efd5
|   |-- <Hash> 1.0
|-- <ESPAsyncWifiManager> 0.19.0+sha.1c02154
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP Async WebServer> 1.2.0+sha.95dedf7
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.9.1+sha.3df4efd5
|-- <ArduinoJson> 6.9.1+sha.3df4efd5
|-- <NeoPixelBus> 2.5.0+sha.9619fef
|   |-- <SPI> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
[... 156 lines cut ...]
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12onDisconnectESt8functionIFvPvPS_EES1_+0x4): multiple definition of `AsyncClient::onDisconnect(std::function<void (void*, AsyncClient*)>, void*)'; .pio\build\d1_mini\libb9c\libESPAsyncTCP@src-66213e8e5399ee92a319916838e73378.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12onDisconnectESt8functionIFvPvPS_EES1_+0x4): first defined here
c:/users/lenovo/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\liba0f\libESPAsyncTCP.a(ESPAsyncTCP.cpp.o): in function `AsyncClient::onPoll(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onPollESt8functionIFvPvPS_EES1_+0x4): multiple definition of `AsyncClient::onPoll(std::function<void (void*, AsyncClient*)>, void*)'; .pio\build\d1_mini\libb9c\libESPAsyncTCP@src-66213e8e5399ee92a319916838e73378.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onPollESt8functionIFvPvPS_EES1_+0x4): first defined here
c:/users/lenovo/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\liba0f\libESPAsyncTCP.a(ESPAsyncTCP.cpp.o): in function `AsyncServer::onClient(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncServer8onClientESt8functionIFvPvP11AsyncClientEES1_+0x4): multiple definition of `AsyncServer::onClient(std::function<void (void*, AsyncClient*)>, void*)'; .pio\build\d1_mini\libb9c\libESPAsyncTCP@src-66213e8e5399ee92a319916838e73378.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncServer8onClientESt8functionIFvPvP11AsyncClientEES1_+0x4): first defined here
c:/users/lenovo/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\liba0f\libESPAsyncTCP.a(ESPAsyncTCP.cpp.o): in function `AsyncClient::onAck(std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient5onAckESt8functionIFvPvPS_jjEES1_+0x8): multiple definition of `AsyncClient::onAck(std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>, void*)'; .pio\build\d1_mini\libb9c\libESPAsyncTCP@src-66213e8e5399ee92a319916838e73378.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient5onAckESt8functionIFvPvPS_jjEES1_+0x8): first defined here
[... cut ...]
========================================================== [FAILED] Took 16.95 seconds ==========================================================

Environment    Status    Duration
-------------  --------  ------------
d1_mini        FAILED    00:00:16.949
===================================================== 1 failed, 0 succeeded in 00:00:16.949 ===================================================== 
The terminal process "C:\Users\lenovo\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'd1_mini'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.```

leoncoolmoon, 2021-11-18:
fixed it by mark off 
```
   ; https://github.com/me-no-dev/ESPAsyncTCP#7e9ed22
   ; https://github.com/me-no-dev/AsyncTCP#90715ae6
```
in the "lib_deps =" of platformio.ini
as per:
https://github.com/esphome/issues/issues/683#issuecomment-534704906

closed 2022-04-26 by probonopd