probonopd/WirelessPrinting #176: Don't print in Artillery genius pro

aacorvalan, 2022-05-20:
Hello, in artillery genius it can be connected to the Tx and Rx of the TFT? because I have it connected and it does not print
[image]
[image]

probonopd, 2022-05-20:
Which firmware is the Artillery Genius Pro using? Marlin? Which version?

probonopd, 2022-05-20:
What do you see if you connect via Telnet?

aacorvalan, 2022-05-20:
The latest updates of TFT and Marlin. Telnet with what command?
[image]
[image]
[image]

probonopd, 2022-05-21:
You do not need to specify a port to telnet, it will work with just the IP address.

But there is no communication going on between the printer and the module according to the empty telnet screen. Hence please triple check your wiring. TX on the ESP must go to RX on the printer and vice versa.

aacorvalan, 2022-05-25:
I cross-connected Tx and Rx on the TFT, supposedly they are connected in parallel with the motherboard, with ESPcam and with Nodemcu. With Esp3d I can move motors but not transfer Gcode, that is, the communication worked, I don't know why in this case it didn't