probonopd/WirelessPrinting #180: SD card connected via SPI to ESP32dev - is not recognized

octavius08, 2022-11-21:
First of all good work, this is the first firmware that allows a wifi interface to upload **and** print files.

However, I have a problem with my SD card, it is not recognized. I connected the SD card as follows (SPI standard pins). The 32GB SD card is also formatted under linux vfat.

my hardware:
Creality3D v1.1.3  ATMEGA1284P  8bit
Marlin 2019-07-05  bugfix-2.0.x

SDcard	=>	ESP32
3V3		=>	3.3V
CS		=>	GPIO 5
MOSI	=>	GPIO 23
SCK/CLK	=>	GPIO 18
MISO	=>	GPIO 19
GND		=>	GND

[image]

Info form the UI WirelessPrinting
```
Free heap: 229688

File system: SPIFFS
Filename length limit: 11
Uploaded file: /cached.gco
Uploaded file size: 1136781

Last command sent: M105
Last received response: ok T:65.96 /0.00 B:44.15 /0.00 @:0 B@:0

EXTRUDER_COUNT: 1
AUTOREPORT_TEMP: false
PROGRESS: false
BUILD_PERCENT: false
```

does anyone have an idea how i can activate the sd card, maybe i should use the GPIO connection like the TTGO-T1? thank you in advance for your support.

SDcard	=>	ESP32
3V3		=>	3.3V
CS		=>	GPIO 14
MOSI	=>	GPIO 15
SCK/CLK	=>	GPIO 13
MISO	=>	GPIO 2
GND		=>	GND

[image]

probonopd, 2022-11-21:
Do you see communication going on when you log in using telnet?

octavius08, 2022-11-22:
Many thanks for your response, on telenet i just see this during upload and print,: 

`<ok T:18.75 /0.00 B:17.62 /0.00 @:0 B@:0#ok#
>M117 Receiving...
<ok#ok#
>M117 Received
<ok#ok#
>M300 S500 P50
<ok#ok#
>M117 Printing...
<ok#ok#
>M300 S500 P50
<ok#ok#
>G90
<ok#ok#
>G28
<echo:busy: processing#busy#
<echo:busy: processing#busy#
`.........
it was a simple bed leveling gcode.

I can see via telenet that various functions are running. I can't see whether there is communication with the SD card, M20 only outputs the following code:

[image]

Info form the UI WirelessPrinting it shows flash memory SPIFFS

Free heap: 229688
File system: **SPIFFS**
Filename length limit: 11
Uploaded file: /cached.gco
Uploaded file size: 1136781

Which GPIOs should I use: the standard SPI GPIOs or those of the TTGO-T1? The SPI standard connection works with an SD test under ArduinoIDE?

octavius08, 2022-11-22:
I did a little test with your gcode to check the SD connection, this is the result:

**The gcode:**
`M22 ; Release SD card
M21 ; Init SD card
M928 log.gcode ; Start logging to SD card
`

**The result**
[image]

But I believe the M20 is trying to connect to the memory card on the Ender board and not to the SD via SPI on the ESP32 board. During the test, however, I did not have a card in the SD slot on the Ender Board.