probonopd/WirelessPrinting #182: SD Pinout?

kleffa, 2024-05-28:
anyone have pinout to conect SD to boards or it can be setup by thae code?
Im trying to flah it to a esp32cam, wich comes with a SD card slot, but not sure if pins are already

probonopd, 2024-05-30:
Good idea to use the esp32cam due to its built-in SD slot. I don't think I have tried using it so far.

Probably you would have to adjust 

https://github.com/probonopd/WirelessPrinting/blob/1f4a381489fd458cd358db380c1d058e162b4706/ESP8266WirelessPrintAsync/StorageFS.h#L20

to match the esp32cam.

kleffa, 2024-05-30:
it should be         
SPI.begin(14, 12, 13, 15); // esp32cam built in microSD

but I have difficulties to compile it because libraries, have a pack of them? it compiles from arduino ide?

probonopd, 2024-06-02:
Actually now I vaguely remember that I had looked at this some time ago and had concluded that the pinout of th ESP32Cam wasn't compatible. Don't remember the details though, and I might be wrong, so please take with a grain of salt. Anyhow, I don't know how to make it work on this board.

kleffa, 2024-06-02:
I read back some issues, I think problem was that serial UART0 is at GPIO 1
and 3, but never planted what if setup serial after and set to pin 0 and 16

probonopd, 2024-06-02:
So do we agree that it is not possible with this hardware?